OpenRoberta/openroberta-lab #1054: You cant Expand only the Start block

heini208, 2021-07-01:
**Describe the bug**
 - when having a Program the only way to collapse the start block is by collapsing the whole program some goes the other way around, when having a collapsed program you cant expand only the Start block but have to expand the whole program.
 This is especially impactful when having an Arduino program since the whole program is inside a block that's treated similar to the start block
**To Reproduce**
Steps to reproduce the behavior:
1. Have a Program of any size preferably with some variables
2. try to collapse only the start block
3. only help appears no collapsing option
4. then right click an empty space
5. click Collapse blocks
6. try to open only the start block
7. only help appears

**Expected behavior**
 - the start block should have a collapse/expand button

**Screenshots**
[image]
[image]

**Device information**
 - Type Desktop
 - OS: Linux Ubuntu 18.04
 - Browser Chrom/Firefox was tested

bjost2s, 2024-10-24:
depends on old blockly, no time to work on

closed 2024-10-24 by bjost2s

PamoCurtis, 2024-10-24:
Can i close my PR nr.69 in the blockly repo?