OpenRoberta/openroberta-lab #1060: Variables can be used out-of-scope without triggering an error when generating the code

PamoCurtis, 2021-07-08:
### Problem
- Initializing/declaring variables through **for**-, **foreach**- and **function**-blocks and using the "**set 'variable' to**"-block out of scope, does not trigger an error when opening the codeview/simulation/sourceCodeEditor (**pic 1&2**).
- Initializing/declaring variables like above but then deleting the block lets you use the variables even though they are never initialized/declared (**pic 3)
- It is also possible to use the **variable**-block after the initialization or in another function. In both cases the variables are out of scope (**pic 4**)
- It is even possible to run the code and debug it (**pic 2**). 
- The code that is generated is faulty 

### Expected behavior
 - Triggering anything that generates the code, the codeview/simulation/sourceCodeEditor should not be opened and an error message should appear like in **picture 5** (picture is from the [open-roberta lab](https://lab.open-roberta.org))

### Screenshots
**PIC 1**
[image]

**PIC 2**
[image]

**PIC 3**
[image]

**PIC 4**
[image]

**PIC 5**
[image]

PamoCurtis, 2022-05-04:
Bug is still present in the lab and test/dev

RaghuvirShirodkar, 2023-03-03:
Almost all issues* listed in the description seem to be resolved after testing on test server.

*For issue listed in pic 3 above, the set var block still remains in the program after removing the corresponding block that initialises the var. This issue is tracked in #1233.

bjost2s, 2023-03-08:
#1233 is solved, too

closed 2023-03-08 by bjost2s