OpenRoberta/openroberta-lab #1063: Opening the Feedback menu - Uncaught TypeError

aykborstelmann, 2021-07-15:
Open the feedback menu in the lab, open the developer console.
See ` Uncaught TypeError: Cannot read property 'initialise' of undefined ` in the console.

I seems that we are trying to initialize cookies here.

Akshay1810, 2021-07-24:
Please provide link of that page I'm not able to find it.

aykborstelmann, 2021-08-06:
[image]

This is the last icon on the sidebar menu.

But since some resources in the feedback menu are from our roberta hompage (https://www.roberta-home.de/) we prefer to resolve this error by ourselves.

bjost2s, 2022-05-03:
A solution will be to integrate the form in the lab's frontend code and not to use www.roberta-home.de