OpenRoberta/openroberta-lab #1071: mBot: faulty config leads to server-error

PamoCurtis, 2021-07-22:
### Problem
modifying the motors in the mBot config like e.g in **PIC 1** and using any of the drive or motor blocks, unless the "stop motor port mX" block, results in a server-error when opening the code-view (**PIC 2**)

### Expected
Error message saying "There are errors in your program or configuration. Please check the messages." with a valid error message on the block itself.

**PIC 1**
[image]

**PIC 2**
[image]

rishabhdeepsingh, 2021-09-08:
Can we close this one now? @aykborstelmann @rbudde

aykborstelmann, 2021-09-08:
We usually wait until the changes are released until we close the issues. For now we just add the label resolved and move it to done in our project

rishabhdeepsingh, 2021-09-08:
Cool, Thanks for the info.

bjost2s, 2022-01-04:
The solution is not sufficient:
[image]
There is still a server error:
[image]

and the drive blocks are not checked at all, see:
[image]
with no motor in the configuration

rishabhdeepsingh, 2022-01-05:
This solves this error.

closed 2022-01-08 by bjost2s