OpenRoberta/openroberta-lab #1106: Importinga program fails sometimes

rbudde, 2021-09-10:
when importing a program, sometimes, but reproducable, the import silently fails. I found the following error in the logging:

```
162 | [[ERR ]] | [[ERR ]] 29 msec: open program 'messages' from XML, then EXCEPTION: TypeError: Cannot read properties of
undefined (reading 'length') with stacktrace: Error at Object.wrap (https://test.open-roberta.org/js/helper/wrap.js:37:27) at 
Object.rest [as success] (https://test.open-roberta.org/js/helper/wrap.js:95:20) at u (https://test.open-
roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:27457) at Object.fireWith [as resolveWith] (https://test.open-
roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:28202) at k (https://test.open-roberta.org/js/libs/jquery/jquery-
3.3.1.min.js:2:77651) at XMLHttpRequest. (https://test.open-roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:79907)

-- | -- | --
161 | [[ERR ]] | Cannot read properties of undefined (reading 'length')

```
seems to be a FATAL problem. It may depend on the size of the program (they are large)

rbudde, 2021-11-23:
I could reproduce the error:

1. select Calliope
2. load the program (windows path ): `OpenRobertaServer\src\test\resources\crossCompilerTests\robotSpecific\calliope\sensors_all_without_pins_and_callibot.xml` 
3. load the program `OpenRobertaServer\src\test\resources\crossCompilerTests\robotSpecific\calliope\sensors_pins.xml`
4. the second load fails, the first program remains loaded and the log of the lab (menu help -> logging) shows the error

RaghuvirShirodkar, 2021-12-02:
It seems the error stems from specific blocks used in conjunction with `wait until`. I've tested the program `[...]\crossCompilerTests\robotSpecific\calliope\sensors_all_without_pins_and_callibot.xml`, since it's the one for Calliope that has this problem.

I'm attaching the [following files (zipped)](https://github.com/OpenRoberta/openroberta-lab/files/7641560/Files.for.testing.zip) to check reproducibility of error with and without specific blocks:
* `sensors_all_without_pins_and_callibot.xml` => the original file.
* `calliope-sensors-2-with-ultrasonic-humidity-sensors.xml` => updates to original file, retaining only ultrasonic and humidity sensors in `sensorsWaitUntil` function.
* `breaking-calliope-sensors-2-with-ultrasonic-humidity-accel-sensors.xml` => similar to updated file, adding accelerometer.

In order to test it out, follow these steps:
* Start server locally (or lab.open-roberta.org), select Calliope as the robot and import original file `sensors_all_without_pins_and_callibot.xml`.
* Try to import another file (any), the import will fail and the log will show
```
[[ERR ]] | [[ERR  ]] 47 msec: open program 'calliope-sensors-2-without-other-sensors'  from XML, then EXCEPTION: TypeError: e[f].menuGenerator_ is undefined  with stacktrace:  
wrap@https://lab.open-roberta.org/js/helper/wrap.js:37:27 
rest@https://lab.open-roberta.org/js/helper/wrap.js:95:20 
u@https://lab.open-roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:27457 
fireWith@https://lab.open-roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:28202 
k@https://lab.open-roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:77651 
n/<@https://lab.open-roberta.org/js/libs/jquery/jquery-3.3.1.min.js:2:79907
-- | --
[[ERR ]] | e[f].menuGenerator_ is undefined
```
---
* Refresh page, select Calliope as the robot and import updated file `calliope-sensors-2-with-ultrasonic-humidity-sensors.xml`.
* Try to import another file (any), the import will succeed.
---
* Refresh page, select Calliope as the robot and import updated file `breaking-calliope-sensors-2-with-ultrasonic-humidity-accel-sensors.xml`.
* Try to import another file (any), the import will fail with the same error as before.

The only difference between the non-breaking and breaking files is the presence of a `wait until` block with accelerometer (in this case). The problem is also encountered with other sensors, namely gyroscope, light and temperature sensors.

rbudde, 2021-12-02:
thank you for the examples. Definitively these examples show, that the program _in the blockly workspace_ effect the error. The simplest case I found, was: 

- select Calliope
- export a program with only the start block
- write the program as attached
[image]
- import the program with only the start block
- this fails, see the error in the log

This simple setup could guide us to find the error

closed 2022-01-10 by bjost2s