OpenRoberta/openroberta-lab #1108: Issue occured while Installation on ubuntu(MobaXterm)

KondraDivya, 2021-09-14:
I installed everything successfully about robots on ubuntu and step 1 is successfully done while coming to step 2 that is the creation of database there it is coming an issue

[image]

I

aykborstelmann, 2021-09-16:
Did you try `./admin.sh git-mode create-empty-db` (git-mode without starting with a hyphen) ?

KondraDivya, 2021-09-17:
Yes I tried it by removing hyphen still getting an issue as follows:
[image]

rbudde, 2021-09-17:
sorry for answering late. I'm the author of these scripts. `-git-mode` is correct, it's a convenience parameter.

The most likely reason is, that you didn't run `mvn clean install` (or `mvn clean install -DskipTests` to save some time) before. The mvn call installs -among many other things- the database jar needed for the operation. Can you check this?

bjost2s, 2022-01-27:
please reopen if the problem still exists

closed 2022-01-27 by bjost2s