OpenRoberta/openroberta-lab #1109: EV3dev steering backwards does not move the robot backwards

Ch40sl0rd, 2021-09-14:
when using the "steer backwards" block, either with or without specifying a distance to drive, the physical robot moves forwards. Only when using a negative distance does the robot actually drive backwards.

Ch40sl0rd, 2021-09-14:
@ensonic could you please take a look at this, I assume the problem lies in the method DriveInCurve in ev3.py

PamoCurtis, 2021-11-24:
Is this issue still existing? Because i can't reproduce it on **lab.open-roberta.org** nor **test.open-roberta.org**

bjost2s, 2022-01-27:
please test this again and share the results

RaghuvirShirodkar, 2022-02-11:
It seems that the latest EV3dev package ([release 1.7.4](https://github.com/OpenRoberta/robertalab-ev3dev/releases/tag/v1.7.4)) has a typo in `driveInCurve()` method, as can be [seen here](https://github.com/OpenRoberta/robertalab-ev3dev/blob/9100fae26830e2b0230fa81d10053854d1951406/roberta/ev3.py#L520).

As a result, the robot never `steers` backward, because the code expects parameter backward***s***!

Furthermore, it seems release 1.7.4 (dated 19.01.2018) is [missing commits](https://github.com/OpenRoberta/robertalab-ev3dev/compare/v1.7.4...develop) made to `develop` since the release. A quick look at the commits shows that the corrections are already in `develop`, so perhaps a new version of the package needs to be released?

PR #1224 (2022-03-07): Issue #1109: fix steer block for ev3dev with helper methods.
# Description

This PR implements the following changes:
- add hal's driveInCurve() implementation to ev3dev.methods.yml.
- update relevant visitors and workers (**Ev3PythonVisitor**, **DifferentialMotorValidatorAndCollectorVisitorEv3**, **Ev3ValidatorAndCollectorWorker**).
- update crossCompilerTests for ev3dev.

Fixes #1109 

## Type of change

Please delete options that are not relevant.

- [X] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

- Tested on the robot (ev3dev). 
- Updated crossCompilerT [... cut]
closed 2022-03-07 by bjost2s