OpenRoberta/openroberta-lab #1119: Open Roberta Lab: No audio output directly possible from within Calliope mini app

joernalraun, 2021-09-27:
When using the Calliope mini app (iOS) there is no way to directly hear audio blocks through the simulator.

The only way to activate it (though this seems not to work throughout all devices/iOS versions) is to close the app, start the Lab in the browser, allow microphone access and open the app again.
This is a long way to get the sound activated.

bjost2s, 2021-09-27:
Probably an issue of the app itself? [Here](https://stackoverflow.com/questions/33496883/html5-audio-not-playing-on-uiwebview-ios) is a similar problem description.

joernalraun, 2021-09-29:
Good point, I'll check if this could be a solution.
I think this can be closed for now!

closed 2021-10-02 by bjost2s