OpenRoberta/openroberta-lab #1123: Overwrite of robot plugin properties at runtime

rbudde, 2021-10-05:
the main properties of the lab are stored in `openroberta.properties` . These properties can be overwritten at startup time of the server by command line parameters `-D <prop-key>=<prop-value>` .

The same mechanism should be available for property files for robot plugins, e.g. `ev3.properties` . The command line parameter could be
* `-R <robot> <prop-key>=<prop-value>` or
* `-R <robot>.<prop-key>=<prop-value>`

Changes are 'close' to the startup class `ServerStarter `.

RaghuvirShirodkar, 2021-11-18:
The code already supports this, albeit through a different command line parameter.

In `ServerStarter.java`, option parser accepts both `-d` (for overwriting server properties) and `-D` (for overwriting robot properties).

A `LOG.info` of `basicPluginProperites` in `Util::configureRobotPlugin` shows that the command line arguments are indeed overwriting the robot properties:
Command line arguments passed:
```bash
-d server.log.level=INFO -D microbit:robot.real.name="new-microbit" -D nao:robot.real.name="new-nao"
```

Log output:
```bash
2021-11-18 12:03:57,198 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util - new plugin property from command line: microbit:robot.real.name=new-microbit
2021-11-18 12:03:57,199 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util - Before: micro:bit
2021-11-18 12:03:57,199 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util - After: new-microbit
2021-11-18 12:03:57,306 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util - new plugin property from command line: nao:robot.real.name=new-nao
2021-11-18 12:03:57,307 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util - Before: NAO
2021-11-18 12:03:57,307 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util - After: new-nao
```

The format expected is:
```bash
-D <robot-name>:<robot-property-key>=<robot-property-value>
```

Should the command line argument `-D` be updated to `-R`?

rbudde, 2021-11-18:
thank you very much! Iimplemeted his a long time(?) before and forgot it :-) The parameter name can stay as it is.
The issue doesn't need a further review)

closed 2022-01-10 by bjost2s