OpenRoberta/openroberta-lab #1160: Obstacle can not deleted within sim

Thorlogy, 2021-12-06:
**Describe the bug**
 - Within the simulation a delete the standard obstacle after closing and obining the sim window the obstacle will be displayed again
 - 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'SIM"
2. Delete the obstacle
3. Click on SIM (close sim)
4. Click on SIM again (open sim again)
5. See error

**Expected behavior**
 - Deletion of the standard obstacle will last until reload of the or lab
  
**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type [Desktop/tablet/smartphone]
 - OS: [e.g. windows], including version
 - Browser [e.g. chrome, firefox], including version

**Additional context**
 - Add any other context about the problem here.

badgujarsachin83, 2021-12-07:
i don't understand what is issue. can you please reproduce this issue with video format. it is better to understand

Thorlogy, 2021-12-07:
Please try yourself. Go to OR Sim - delete the blue object. Close the Sim. Open the Sim again. And you will see the object is there again. 

[image]

rbudde, 2022-05-04:
works as expected

closed 2022-05-10 by bjost2s