OpenRoberta/openroberta-lab #1161: Reminder of the starting point

Thorlogy, 2021-12-06:
**Is your feature request related to a problem? Please describe.**
 - If i would like to start the robot in the or sim at a specific place/spot i have to place the robot by hand (again and again)
 
**Describe the solution you'd like**
 - I place the robot in the or sim to a specific spot and click on "remember" - then a can easily place the robot to this selected spot

**Describe alternatives you've considered**
 - A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
 - Add any other context or screenshots or a mockup about the feature request here.

rbudde, 2021-12-07:
Please remove dead text in the description.

The request is implemented to 50% already. If you save the configuration of obstacles etc, the robot position is saved, too. On reload the robot is positioned as requested (the existing 50% :-)

What remains to be implemented: when the reset position button is pressed, THIS position has to be used (the missing 50% :-)

Thorlogy, 2021-12-07:
Thx. 

But I could imagine that the handling is a bit confusing. If I have understood it correctly:  
1. configuration of the simulation environment incl. positioning of the robot. 
2. download the self-configured simulation environment. 
3. upload this simulation environment.
Correct? 

Wouldn't it be more understandable if the robot position could (additionally) be saved directly? Without having to download the entire environment and upload it again?

rbudde, 2021-12-07:
Imagine: the first 2 steps are done by a teacher, who sends the conf of the sim environment to his/her pupils. They load it and use it. Clicking on the position button sets the correct position over and over again.

If you want to save the position of the robot INTERNALLY (i.e. without exporting it to some conf file), we would need another button (e.g. "set robot default position"). It would not be possible to share that.

Two possibilities, that could be implemented.

Thorlogy, 2021-12-07:
You're right. For the first idea with the teacher, this fits perfectly! 

For implementation or for testing of own ideas (i.e. until you have a solution that you can share) it is a bit uncomfortable. Another button would be very useful for this - I think.

bjost2s, 2022-08-25:
solved with the "new" simulation

closed 2022-08-25 by bjost2s