OpenRoberta/openroberta-lab #1164: Cannot reset my password anymore - server does not send mails

bjost2s, 2021-12-07:
When I try to reset my passwort and type in the correct email I got the following message
[image]
and no mail

rbudde, 2021-12-07:
On sunday we updated/upgraded the ubuntu-18.04 of out prod machine. Now it has the same level as our test machine. This may be the cause of the problem.

But: Mail to internal and external adresses works fine with the test server.

rbudde, 2021-12-13:
The upgrade of Ubuntu lead to a restriction in name resolution. The name of the postfix server `me-roblab-prod` could nor be resolved anymore. Changes in the start parameters of the server are needed: Instead of `me-roblab-prod` the ip address of the gateway of the docker network `ora-net` (that is `172.20.0.1` on the prod machine) has to be used

bjost2s, 2022-01-18:
fixed with changed mail properties

closed 2022-01-18 by bjost2s