OpenRoberta/openroberta-lab #1186: Buttons disappear in other sim after opening Nao sim

PamoCurtis, 2022-01-06:
# **The Bug and how to Reproduce**
Just like the title says, system-view and sensor-view buttons disappear in other (debug) simulations after opening the NAO simulation 
- Select Nao robot
- Open simulation
- Close simulation
- Switch to any other robot that has a simulation (EV3, NXT, Microbit, Calliope, Mbot)
- Open (debug) simulation
- Buttons like sensor-view, system-view are not present

**Usual EV3 buttons in the simulation**
[image]
**and debug simulation**
[image]

**EV3 buttons after opening the Nao simulation and switching to the EV3 simulation**
[image]
**and debug simulation**
[image]

PamoCurtis, 2022-03-26:
There was one more problem: when having the microbit/calliope simulation started and switching to the Nao-simulation, the resetPose-button was missing, this is also fixed fixed with my PR.

rbudde, 2022-05-04:
works as expected

closed 2022-05-10 by bjost2s