OpenRoberta/openroberta-lab #1191: Function names can cause infinite recursion

heini208, 2022-01-07:
**Describe the bug**
 - Most robots generate a function like main() or run() in their code generation regardless of blocks used
 - The problem is these names are possible custom function names which will then cause infinite recursion in languages like python and errors in languages like java 
 - e.g. you make a function named main() and use it in a robot using python your program would now look like this:
 EV3 example:
[image]

In python inside run() the last main method would be used which is the normal main method which would then call run which would then call main again causing an infinite loop and the custom function to be ignored

**To Reproduce**
Steps to reproduce the behavior:
1. create a program that uses a function called main() or run() in any robot generating these functions
2. try the generated function structure in an online compiler or the actual robot
4. See error

**Expected behavior**
 - the automaticcaly generated function names should either be called _run and _main
 - or functions should not be able to be named main or run

**Additional context**
 - in java programs the name main is fine due to overloading and it beeing impossible for a user to create a method `void main (String[] args)` however run() beeing used twice would not compile

bjost2s, 2022-08-25:
It is recommended to fix this first in blockly. There we have for each programming language a list of illegal names.

rbudde, 2022-09-21:
solved

closed 2022-09-21 by rbudde