OpenRoberta/openroberta-lab #1208: Logging of cross compiler erros should not log error caused by making mistakes in the source code editor

heini208, 2022-01-24:
**Describe the bug**
 - When somone writes invalid code in the source code editor and then tries to compile it, the compilation error is added to the cc-error logging this makes finding new mistakes caused by bugs tedious.  
- the majority of cc-errors logged are most likely due to the source code editor

**Expected behavior**
 - compiler errors should not be added to the logs if the source-code editor was used
 - in other words only code generated by NEPO should be logged.

rbtnx, 2022-02-01:
I'm currently looking into this issue. The cc-error log message is created by the CompilerWorkers for the different systems. Would filtering out all failed compilations logged with
```java
LOG.error("compile {} program {} failed with {}", robot, programName, workflowResult);
```
solve this? Do only source code editor compilation errors get logged here or also compilation errors resulting from wrong NEPO generated code?

rbudde, 2022-02-02:
We want to distinguish two cases:
- we generate source code from blockly. In this case, If a compiler error occurs, we have implemented something wrong. We need the logging to find the error
- the user used the source code editor and wrote an incorrect program. Of course, he/she should see the error message (it's shown ugly, b.t.w.), but we need NO logging.

Thus in the context of the lines which log the error, it must be known, what is compiled (our generated code or user edited code). This is known in class `Project` (look for the line with the comment `Used to run native code directly`). I would add a method (there are some different implementation possible), `Project#isUserWrittenCode` (name can be better, of course) and suppress the logging using this.

rbtnx, 2022-02-02:
I noticed that the cc error gets logged twice:
  * first in level `INFO` as defined by the `crosscompiler_error` logger in `logback.xml`
  * then again in level `ERROR` because the workflow hasn't succeeded.
  
Is this intended or can I change the `crosscampiler_error` loglevel to `ERROR`
I tested an implementation of the log filter as @rbudde suggested and it works (when also changing the loglevel as described)

rbudde, 2022-02-03:
the `crosscompiler_error` logger has been introduced, because we want
- to see the program source to analyse the code (this is lengthy), when we generated wrong code
- NOT to flood the default logger with this huge amount of text
Thus: two logging are ok, but the one for the default logger should be short :-) and can also occur if user edited code fails (because it is short :-)

rbtnx, 2022-02-11:
Thanks for the feedback, I'll look into it on the weekend :)

closed 2022-03-01 by bjost2s