OpenRoberta/openroberta-lab #1211: Github action "Build changelog" is not working anymore

bjost2s, 2022-01-24:
compare latest releases [here](https://github.com/OpenRoberta/openroberta-lab/releases), seems that [ORA-4.2.0](https://github.com/OpenRoberta/openroberta-lab/releases/tag/ORA-4.2.0) was the last working one (ore only one?)

aykborstelmann, 2022-01-27:
Oh okay, I will have a look at this

aykborstelmann, 2022-01-27:
This should now be resolved. The reason was, that the job was looking for commits containing `issue #[Some Number]`. Actually I already thought about this before and fixed it, but forgot to build the index.js which is executed.

bjost2s, 2022-01-27:
Thank you very much! I have looked for this script, but couldn't find it. Could you give me a hint? Another question, can I run it by hand to update passed releases?

aykborstelmann, 2022-01-27:
Yes, you can find the code here: https://github.com/OpenRoberta/gh-actions-issue-changelog
> can I run it by hand to update passed releases?

Well, that's more difficult. You can actually run it by cloning it and then `npm install && node dist/index.js` or when you need to rebuild and run it `npm run all && node dist/index.js`. The problem is, that is 
1. it does actually updates the release notes, it just outputs it (and the github action in the main repo just use this output)
2. it always uses the last two tags to determine which commits are new for this release. Therefore, you can just run it for the last release, or otherwise would have to change the code ([this is the line](https://github.com/OpenRoberta/gh-actions-issue-changelog/blob/6fcf60bf705927d4ad6ad9471346806a2f21ec40/src/changelog.ts#L6))

If you want to run it locally you also have to set the following environment variables:
* `GITHUB_TOKEN` - this can be generated in the Github Developer Settings
* `GITHUB_REPOSITORY` - this should be `OpenRoberta/openroberta-lab` 

The easier way would be to rerun the Release Action. This would again only work for the latest release because of the way the action works. But on the other hand, this would probably create a new release.

closed 2022-03-01 by bjost2s