OpenRoberta/openroberta-lab #1215: Test/run source-code within OR Simulation

Thorlogy, 2022-01-27:
**Is your feature request related to a problem? Please describe.**
 - for the ev3 (and other systems), it would be nice to have the ability to run modify the source code (c, python, java...) within the simulator

**Describe the solution you'd like**
 - Include a additonal button "Sim" which open the OR Simulation

bjost2s, 2022-01-27:
Unfortunately this is impossible

closed 2022-01-27 by bjost2s

Thorlogy, 2022-01-27:
why?

rbudde, 2022-01-27:
the generator of code for the simulation takes the blockly XML and translates it to some byte code, that is executed by the byte code interpreter of the simulation. Our simulation has no knowledge about the target language of the *real* robot (Python + Java for ev3, C++ for Calliope). If we get Python code edited by the user, for instance, what should we do? Program a full Python interpreter? Get Python running in the Javascript engine of a browser? the same for Java and C++? This is impossible for the team.