OpenRoberta/openroberta-lab #1233: loop-variables aren't automatically deleted when removing the forLoop-block

PamoCurtis, 2022-03-17:
### Describe the bug
2 problems:
- using a loop-variable out of scope and then removing the initial forLoop-block does not remove the loop-variable
- declaring at least two variables so that you can switch between them in the variableSet/Get-block, then removing the forLoop-block, you are still able to switch between the variable name from the removed variable back and forth as long as at least one variable is still named after the removed one

### To Reproduce
Concentrating on the second problem because that includes the first one too:
1. Create two forLoop-blocks ( or one forLoop-block and a global variable, that doesnt matter)
[image]

2. Create two or more variable-blocks, select the variable name of the forLoop-block and use it out of scope (e.g below the forLoop-block)
[image]

3. Remove the forLoop-block
4. See error (you can even open the code-view and its generates the code even though it shouldn't)
[image]

### Expected behavior
 - The loop-variable should get deleted automatically when removing the forLoop-block, like it happens when removing the declaration of a global-variable
 
### Example Program
[example.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/8284399/example.xml.txt)

PamoCurtis, 2022-03-21:
Ok, same happens when declaring a variable in a function-block, using a variables_set/get block out of scope and deleting the function-block

*before deletion*
[image]

*after deletion*
[image]

PamoCurtis, 2022-05-04:
Bug is still present

RaghuvirShirodkar, 2023-03-03:
Issue still exists on test server.

bjost2s, 2023-03-08:
Solved