OpenRoberta/openroberta-lab #1266: Small Changes - User Interface. Behavior

rbudde, 2022-04-07:
Change requests:
- [ ] TO_BE_DISCUSSED: weights and bias are shown in floating point mode. Show fix number of digits (should this be user-defined?)
- [ ] TO_BE_DISCUSSED: thickness of arcs should be configurable between 5 and 10. Or set to 6, e.g.
- [ ] ...

This ticket can be open for some time and should assemble small requests (~1h time for implementation).
From time to time it can be closed and reopened with an empty feature list.

**For new requests edit this issue text and prefix your request with TO_BE_DISCUSSED. If it's decided to implement the request, update the request text and remove the prefix.**

Thorlogy, 2022-04-27:
It would be awesome if you could see the mesh during automated training so you could see the changes to the weights (similar to Tensorflow)....

Thorlogy, 2022-04-28:
It would be nice to have a additional block: "get weight from ... to ..."

Thorlogy, 2022-04-28:
With a view to a follow-up project, we should consider how smartphones can also be used for our KNN integration. For pupil could that gain a lot of fun.

Thorlogy, 2022-05-02:
It should be possible to create at least 8 neuron within hidden layer

Thorlogy, 2022-05-03:
Representation of the ANN as a matrix

rbudde, 2022-05-04:
all comments so far moved to the item list

Thorlogy, 2022-05-13:
Numbers with comma - instead of point - are "made" / convert to 0

Thorlogy, 2022-05-18:
It would be nice if the edge weights could also be entered as fractions (1/2 1/3...).

rbudde, 2022-05-30:
all comments so far moved to the item list.

This issue will be closed end of May

closed 2022-05-31 by bjost2s

Thorlogy, 2022-06-17:
the nn input and output Blocks should be pre-named (like variables) and functions

Thorlogy, 2022-06-17:
Note in the tab -Neuronal Networks- : "In order to be able to program neurons here, you have to drag at least one input and one output neuron in the tab -Program NEPO- in the category >Neuronal Network< into the bracket >make a NN step<. 

--> A picture or gif, would be perfect

Thorlogy, 2022-06-20:
for future updates: For the use in school classes it could be important to enable a possibility (interface) to upload files (CSV e.g. *Iris train  or some data set from SenseBox). In addition, it should be possible to display the sensor data display instead of a robot display.  

*https://gist.github.com/netj/8836201
https://github.com/ramp-kits/iris/blob/master/data/train.csv

Thorlogy, 2022-06-20:
Integration of a Boolean activation function

Thorlogy, 2022-06-21:
After clicking on a neuron (status = yellow), you cannot return it to its original status by clicking on it again. You have to click on another neuron or click on the drop-down "show values".

[image]