OpenRoberta/openroberta-lab #1267: wrong/incomplete Error on some blocks with missing parameters

roffinator3000, 2022-04-08:
When trying to view the source code while using a block that does not have all input values satisfied, a general warning and indicators on the concerned blocks should be shown. But if there are missing inputs on the blocks for building texts or generating a (not empty) list, these block do not have the indicator showing the missing input, only the general message shows up

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `dev.open-roberta.org/` and select an EV3
2. Click on '*2' (expert toolbox)
3. Take either `create text with` from `Text` or `list` from `Lists` and integrate them into some executable code
4. Remove one or more inputs of this block
5. Click on `show source` or `simulate`
7. See missing error

**Expected behavior**
 - a circeled X should be shown in the block with missing input
 - a text bubble should appear over the same block, explaining the problem in detail
 - a popup should appear and inform there is some error in the program (this happens correctly)

**Demo code**
[bugDemo_NEPOprog.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/8451982/bugDemo_NEPOprog.xml.txt)
 - the right purple block (`list`) and the left green block (`create text with`) will prompt the error-popup but do not have the explaining bubble or indicating X
 - the lower orange block works correctly

**Device information**
 - Desktop
 - OS: Ubuntu 20.04.4
 - chrome 100.0.4896.75 (64-bit) and firefox 98.0.2 (64-Bit)

bjost2s, 2022-05-02:
somehow related to #1286

PamoCurtis, 2022-05-04:
Tested successful on `test.open-roberta.org`.

closed 2022-05-10 by bjost2s