OpenRoberta/openroberta-lab #1277: Names of projects in gallery appear twice (parts of at least)

joernalraun, 2022-04-14:
It seems that for some reasons projects are displayed with a part of the original name.
All files should be saved the same, though some are displayed different. This might be a feature, but there is no obvious reason why they are shown and if they get the "second name" the original should be hidden.
 
[image]

bjost2s, 2022-05-02:
This is an undocumented feature of the gallery:
The first line of the program's documentation will always be displayed below the name of the program, like in your example.
We should explain this feature in the preview of the gallery tile.

bjost2s, 2023-03-14:
@RaghuvirShirodkar , please create a new issue for the typo, because there is no time to fix it now

closed 2023-03-14 by bjost2s