OpenRoberta/openroberta-lab #1279: Program Export does not work

Thorlogy, 2022-04-19:
With Safari on MacBook MOS Monterey Version12.3.1 the export of a program does not work

joernalraun, 2022-04-25:
This is the same on any MacOS when using Safari.

bjost2s, 2022-04-25:
What exactly happens? What are the error messages if any, or do you see something in the log?

Thorlogy, 2022-04-25:
Nothing happens, no error messages nothing. 

Log: 
[image]

RaghuvirShirodkar, 2022-04-25:
Program export (and import) works on the MacBook that I have here...

## Tested on:
MacOS Monterey v12.1

## Browsers:
Safari v15.2
Firefox v91.5.0esr

Tested on `lab` as well as `test`.

joernalraun, 2022-04-25:
@RaghuvirShirodkar This has been an issue since a couple of years, but only on Safari...

RaghuvirShirodkar, 2022-04-26:
I think I found the bug that causes this issue and how to reproduce it.

## Probable cause
`download` attribute for `a` tag doesn't seem to work for some newer Safari versions.

## Steps to reproduce
1. Go to `lab.open-roberta.org`.
2. Create a sample program for any robot.
3. Click on the _edit_ menu -> _export program_.
4. Observe that nothing is downloaded.

Since I was exporting the program through keyboard shortcuts (Ctrl + E), it was working for me. However, when I try to export the program through the menu, nothing is downloaded.

Thorlogy, 2022-05-04:
Program Export with Safari works now

closed 2022-05-10 by bjost2s