OpenRoberta/openroberta-lab #1285: Calliope mini robot: Button A and B as an selectable option

joernalraun, 2022-04-26:
When working with buttons it would be more intuitive if the dropdown for button A and B would also have the option A and B integrated.

bjost2s, 2022-05-02:
We think that users should use the logical AND to program this behaviour. This might also be a good learning effect.

closed 2022-05-02 by bjost2s

joernalraun, 2022-05-02:
I understand and the approach is close to the way you would code when typing, but when using A, B and A+B with this workaround it is not as easy to use in an application (have a look at "Tasten" from Calliope Code in the gallery on a device), as the execution is not as reliable as you would expect.
Therefore I have seen ways where people build rather complex functions only to make that that button press work as intended.