OpenRoberta/openroberta-lab #1291: Ternary If: wrong code is executed in sim when nested in

PamoCurtis, 2022-05-04:
**Describe the bug**
Executing the example code in the sim will execute both times the false-statement in the first '**ternary if**'-block and then the true-statement in the nested in '**ternary if**'-block even though the first '**ternary if**'-block is already set to true

[image]

**Example Program**
[ternary_if_bug.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/8619539/ternary_if_bug.xml.txt)

rbudde, 2022-05-04:
this is an old error, reproducable on the prod system. The problem is the recursive use of `if/ternary if` structures. The generated code is definitively wrong. The following program has the same problem:
[if_ternary_if_bug.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/8621390/if_ternary_if_bug.xml.txt)

Reason of the problem: the code generator uses a functional pattern to delay the allocation of forward jump locations (because the location is only known after all sub-components have generated code). The jump location is generated too short and executes parts of the code, that should be unused.

rbudde, 2022-05-04:
BTW: Great to find THIS error by writing a test program!

rbudde, 2022-08-15:
seems to be close related to https://github.com/OpenRoberta/openroberta-lab/issues/1340

closed 2022-08-24 by bjost2s