OpenRoberta/openroberta-lab #1292: MBot2 - Integration

bjost2s, 2022-05-04:
tbc ...

- [ ] Check capabilities
- [ ] Adapt compiler workers
- [ ] Add crosscompiler resources
- [ ] Add code generation
- [ ] Add reset worker
- [ ] Add tests
- [ ] Add background image
- [ ] Add icon
- [ ] Add wiki entries/help files
- [ ] Check and support transfer/detection capabilities
- [ ] Add transfer/detection capabilities 
  - [ ] Linux x86
  - [ ] Linux ARM won't be part of this integration, see separate issue #1257
  - [ ] Windows
  - [ ] OSX

heini208, 2022-08-19:
a reset worker is not needed because the mbot2 has multiple program slots and the ones used by the lab are not shared with preinstalled programs.

closed 2022-08-24 by bjost2s

pjfagulha, 2023-04-14:
When is the "simulation visualization" tool in the mBot 2 robot implemented?
When will the beta phase of the mBot2 robot end?
Thank you.

bjost2s, 2023-04-15:
@pjfagulha, thank you very much for your interest,
"When is the "simulation view" tool in the mBot 2 robot implemented?"
Unfortunately, there are no plans yet to support the simulation for mBot2.
"When will the beta phase of the mBot2 robot end?"
As soon as we receive more positive feedback, especially for the newly supported communication blocks.
"When will I be able to transfer the generated code to the robot, without transmission errors?"
Please checkout the latest version from the connector program and report issues here. Our last tests ran without error.