OpenRoberta/openroberta-lab #1293: New / additional math block specialized for touch devices

bjost2s, 2022-05-04:
When working with touch devices, it is particularly necessary when entering letters or numbers that a keyboard appears on the screen, which can then be used for input via touch in the same way as a real keyboard. This process is rather inconvenient and some devices do not even offer the keyboard. In order to improve the input for the user, especially for tutorials, a block is to be developed that enables the input for the math block via dropdown:

[image]

This block has exactly the same properties as the common math block, but some constrains:
- limited to a positiv number from 0 to 999
- integer
and some features:
- you can still type in your number in the upper field, this will also change the values for the dropdowns
- edit the value with the three dropdowns, this will also change the value in the upper field
The block could be fully supported bei the common AST class for the common math block

bjost2s, 2022-05-05:
The desired new programming block should also support negativ values.

rbudde, 2022-05-09:
works in the nao simulation as expected. @bjost2s  Why is the block not available everywhere?

closed 2022-05-10 by bjost2s