OpenRoberta/openroberta-lab #1307: Collection of (very) small changes

rbudde, 2022-05-17:
Change requests:
- [ ] distrribute the nn blocks to the toolboxes. In the beginner: step and input and simple output neuron
- [ ] xxx

This ticket can be open for some time and should assemble small requests (~1h time for implementation).
From time to time it can be closed and reopened with an empty feature list.

**For new requests add comments. I (rbudde) will scan these comments and put them into the list**

Thorlogy, 2022-06-21:
After clicking on a neuron (status = yellow), you cannot return it to its original status by clicking on it again. You have to click on another neuron or click on the drop-down "show values". 

[image]

Thorlogy, 2022-07-04:
Bug?

[image]

wird ein NN Block deaktiviert, funktioniert das Program nicht

Thorlogy, 2022-07-05:
@rbudde 
my a video explains more than i can with words..

https://user-images.githubusercontent.com/19221359/177318859-8d63518f-4a2b-41c7-b6b7-b88a1acd55d2.mov

rbudde, 2022-07-05:
@Thorlogy  I cannot reproduce this. On dev4 the robot works independant of activation/deactivation of the block

Thorlogy, 2022-07-05:
Categorie "Neuronal Network" has no german translation:

[image]

rbudde, 2022-07-19:
**_--------- add NO new change requests here -----------_**

closed 2022-07-29 by bjost2s