OpenRoberta/openroberta-lab #1331: Server error when generating a colour via a function

el-voss, 2022-07-07:
Hello,

I wanted to program a colour wheel in the roberta lab for an Arduino Uno with an RGB LED. For this I coded a function, that takes a number as an argument and gives back a colour. But when I try to use this function in the loop, I get a "server error". The program is attached below. I tried it with firefox and chromium.

Steps to reproduce:
1. Import the program below.
2. Open the code view (will not be opened, but show a server error instead) or press the "play" button.

[farbrad-rgb-led.zip](https://github.com/OpenRoberta/openroberta-lab/files/9062936/farbrad-rgb-led.zip)

RaghuvirShirodkar, 2022-09-26:
This is an issue for arduinos, mbot and sensebox.
There is also a problem when trying to use the following blocks for arduinos:

[image]

Here is the [program](https://lab.open-roberta.org/#loadProgram&&uno&&NEPOprog&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22arduino%22%20xmlversion=%223.1%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22386%22%20y=%22102%22%3E%3Cblock%20type=%22robControls_start_ardu%22%20id=%22l9Lp%60gNd%5DGQJ_0PBfA?z%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cmutation%20declare=%22true%22%3E%3C/mutation%3E%3Cfield%20name=%22DEBUG%22%3E%3C/field%3E%3Cstatement%20name=%22ST%22%3E%3Cblock%20type=%22robGlobalVariables_declare%22%20id=%22Ux8O!kP(#(%7C6%7BiI#;%609;%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22false%22%20declaration_type=%22Array_Colour%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Eitem%3C/field%3E%3Cfield%20name=%22TYPE%22%3EArray_Colour%3C/field%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22robLists_create_with%22%20id=%22@IUt/#.(_gyLS).5f%60C%5B%22%20intask=%22true%22%3E%3Cmutation%20items=%223%22%20list_type=%22Colour%22%3E%3C/mutation%3E%3Cfield%20name=%22LIST_TYPE%22%3EColour%3C/field%3E%3Cvalue%20name=%22ADD0%22%3E%3Cblock%20type=%22robColour_picker%22%20id=%22//S%5D#nl~HUo2%25doyuN-=%22%20intask=%22true%22%3E%3Cfield%20name=%22COLOUR%22%3E#FFFFFF%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD1%22%3E%3Cblock%20type=%22robColour_picker%22%20id=%226%5BJ#8a%5D_t5?Smu2LU_3y%22%20intask=%22true%22%3E%3Cfield%20name=%22COLOUR%22%3E#cc0000%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD2%22%3E%3Cblock%20type=%22robColour_picker%22%20id=%22l,Z%7C:m;!kGXjQ6l:@zkg%22%20intask=%22true%22%3E%3Cfield%20name=%22COLOUR%22%3E#FFFFFF%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/statement%3E%3C/block%3E%3Cblock%20type=%22robControls_loopForever_ardu%22%20id=%22%60mIJB4z:(3D;T%25:%5D?%5EIz%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cstatement%20name=%22DO%22%3E%3Cblock%20type=%22robActions_led_on%22%20id=%22Mj.1fFHw-JGauk0aFSq,%22%20intask=%22true%22%3E%3Cfield%20name=%22ACTORPORT%22%3ER%3C/field%3E%3Cvalue%20name=%22COLOR%22%3E%3Cblock%20type=%22robLists_getIndex%22%20id=%22:A5Mp%5EA:dc:Y5RJ.D5mU%22%20intask=%22true%22%3E%3Cmutation%20statement=%22false%22%20at=%22true%22%20datatype=%22Colour%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EGET%3C/field%3E%3Cfield%20name=%22WHERE%22%3EFROM_START%3C/field%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22robProcedures_callreturn%22%20id=%22%7B1wJ608dy##b%7BQYUGZz~%22%20intask=%22true%22%3E%3Cmutation%20name=%22doSomething%22%20output_type=%22Array_Colour%22%3E%3C/mutation%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22AT%22%3E%3Cblock%20type=%22math_number%22%20id=%22.6P6+uP:E8olMgyS/TVz%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E1%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/statement%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%22386%22%20y=%22329%22%3E%3Cblock%20type=%22robProcedures_defreturn%22%20id=%22CjSV.dh4nbQh%25*CdVxlS%22%20intask=%22true%22%3E%3Cmutation%20declare=%22false%22%20return_type=%22Array_Colour%22%3E%3C/mutation%3E%3Cfield%20name=%22NAME%22%3EdoSomething%3C/field%3E%3Cfield%20name=%22TYPE%22%3EArray_Colour%3C/field%3E%3Crepetitions%3E%3Cvalue%20name=%22RETURN%22%3E%3Cblock%20type=%22variables_get%22%20id=%22Q)s5hKA%7C%7Bm=Wg+9e)dgH%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Array_Colour%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Eitem%3C/field%3E%3C/block%3E%3C/value%3E%3C/repetitions%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/program%3E%3Cconfig%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22arduino%22%20xmlversion=%223.1%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22138%22%20y=%2288%22%3E%3Cblock%20type=%22robConf_led%22%20id=%221%22%20intask=%22true%22%3E%3Cfield%20name=%22NAME%22%3EL%3C/field%3E%3Cfield%20name=%22INPUT%22%3ELED_BUILTIN%3C/field%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%22250%22%20y=%22250%22%3E%3Cblock%20type=%22robConf_robot%22%20id=%22robot%22%20intask=%22false%22%20deletable=%22false%22%3E%3Cfield%20name=%22ROBOT%22%3Eundefined%3C/field%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%2216%22%20y=%22459%22%3E%3Cblock%20type=%22robConf_rgbled%22%20id=%22tO%7B~_Z~E%5DB?l%5B,:H!%5DU=%22%20intask=%22true%22%3E%3Cfield%20name=%22NAME%22%3ER%3C/field%3E%3Cfield%20name=%22RED%22%3E5%3C/field%3E%3Cfield%20name=%22GREEN%22%3E6%3C/field%3E%3Cfield%20name=%22BLUE%22%3E3%3C/field%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/config%3E%3C/export%3E) tested on current `develop`.

The fix needs changes to repo `ardunio-resources` as well, since NEPODefs.h needs a function overload.

closed 2023-02-17 by RaghuvirShirodkar

bjost2s, 2023-03-08:
resolved

PR #1372 (2023-02-17): Issue #1331: fix server error when generating colour with user defined function
# Description

This PR implements the following changes:

- update visitLightAction methods in Arduino, Mbot and Sensebox CPP visitors.
- update tests to accommodate changes.

The relevant changes for `arduino-resources` repo can be found [here](https://github.com/OpenRoberta/arduino-resources/pull/3).

Fixes #1331 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Tested with source code editor. Yet to be tested  [... cut]