OpenRoberta/openroberta-lab #1333: Within "for each item ... in list" "show text" block is not executed as a blocking NEPO block

peppamint, 2022-07-19:
**Describe the bug**
 - Within "for each item ... in list" "show text" block is not executed as a blocking NEPO block

**To Reproduce**
Steps to reproduce the behavior:
1. Import program (attachted)
2. Go to 'Sim'
3. Click on 'Start'
4. Press button B
5. Text 2 will be displayed

**Expected behavior**
 - Expect the content of each list item (0,1,2) to be displayed after pressing button B. 

**Device information**
 - Desktop
 - Windows 10
 - Chrom

** Program **
[Fuer_Wert_aus_der_Liste.txt](https://github.com/OpenRoberta/openroberta-lab/files/9143432/Fuer_Wert_aus_der_Liste.txt)

rbudde, 2022-07-20:
I changed the title, because it is misleading. See the attached program, which shows the same wrong behavior much simpler. The computation of the time for displaying seems to be wrong for short outputs. If you replace the numbers by strings, everything is fine
[zeigeText.txt](https://github.com/OpenRoberta/openroberta-lab/files/9148216/zeigeText.txt)

bjost2s, 2022-07-21:
The reason is that made the blocking depending on "test.length > 0" but test is in this case a number which leads to non blocking because the result in 'undefinded' ;-). Will be fixed together with #907

closed 2022-08-24 by bjost2s