OpenRoberta/openroberta-lab #1348: Propagate renaming of a neuron in the xNN tab to the blockly program

rbudde, 2022-08-15:
Renaming of a neuron is done in the xNN tab. If the renamed neuron is used in the blockly program, then after switching to the program tab the name change should be reflected in the program.

Example: input neuron "n1" is used in the xNN tab. The program uses the block "set 'n1' to value 5". After renaming "n1" to "dist" the program block should read "set 'dist' to value 5".

Implementation idea: in the xNN tab create a list of pairs ('oldName','newName') and use them when the program is displayed. When the xNN tab is opened the next time, the list of pairs should be cleared.

RaghuvirShirodkar, 2023-03-15:
Is this not already implemented?

A snippet from the test server:
[image]

rbudde, 2023-03-15:
after renaming an input neuron in the xNN tab, the following happens: in the program the "old" name is not found and set to the first one found in the list of input neuron names. In your eample you changed the name of the first neuron, thus it seems to work. But if you change the name of the second one, it fails :-<

bjost2s, 2023-05-08:
Tested

closed 2023-05-08 by bjost2s