OpenRoberta/openroberta-lab #1357: NAO Simulation with Chrome + iOS

Thorlogy, 2022-08-29:
**Describe the bug**
 - The NAO Simulation can not be used. The Message is "Safari does not have the technical... Please use a compatible browser (Chrome...) 
 - I use Chrome 

**To Reproduce**
Steps to reproduce the behavior:
1. Use a iOS device + Chrome browser
2. Go to lab.open-roberta.org 
3. Click on NAO 
4. Click on SIM
6. See error

**Expected behavior**
 - NAO Sim works within Chrome browser

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type smartphone / iPad
 - OS: iOS
 - Browser: chrome

RaghuvirShirodkar, 2022-09-20:
It seems `Chrome` as a userAgent is not received in `navigator.userAgent`.

`userAgent` info on iPad 12.9 (BrowserStack):
* Request **Mobile** Site:
  [image]

* Request **Desktop** Site:
  [image]

That being said, when I tried `CriOS` or `Version` for checking for chrome as the userAgent, the simulation reported the following error:
[image]

rbudde, 2022-09-21:
the problem is in software of the WeBots developer. We believe, that the problems are already fixed in a newer version of that software and that we have to upgrade the version we use