OpenRoberta/openroberta-lab #1358: Display the system in use

Thorlogy, 2022-08-29:
**Is your feature request related to a problem? Please describe.**
 - For users it would be helpful if they could see / read on which system their program will run. 
 - So far the message is: "Dein Programm xxx startet gleich auf dem Roboter." even if no robot is used. The english message is better "Your program »NEPOprog« will run in a moment!" 

[image]

**Describe the solution you'd like**
 - Display the name of the system which is used. 

**Describe alternatives you've considered**
 - Use the english version.

rbudde, 2022-09-15:
It is sufficient to remove the german words " auf dem Roboter" from the message. This is sensible for both real robots and simulation.

rbudde, 2022-09-21:
solved

closed 2022-09-21 by rbudde