OpenRoberta/openroberta-lab #1361: Calliope Mini: Using digital read (pulse time) causing light up pixels on the LED display.

MarkusBordihn, 2022-09-04:
**Describe the bug**
If you using digital read (pulse time) for e.g. using an ultra sonic sensor on the Calliope Mini, it will cause light up pixels even if nothing is connect to the calliope mini at all.

**To Reproduce**
Create a program with digital read (pulse time) like the following and run it on the Calliope Mini (even with nothing connected):
[image]
[image]

This will cause the following light up pixels:
[image]

**Expected behavior**
I would expect that using digital read (pulse time) is not causing any light up pixels or other visual side effects.
This could led to wrong conclusions for the users like a broken device, which is not really the case.

**Additional context**
I tested the program without anything connected on two devices including a brand new Calliope Mini with the same result.
Using other programs are not causing any light up pixels at all.

bjost2s, 2022-10-13:
This problem should be compared with Make Code Editor because it  may be a hardware problem

MarkusBordihn, 2022-10-13:
Thanks for the feedback, I was able to workaround this inside the program with the "switch LED matrix" block.

See:
https://github.com/google/calliope-paper-robot/tree/main/hardware/ultrasonic_hc-sr04#e-switch-offon-the-led-matrix

Maybe it's possible to integrate some smart logic in the back-end to automatically use this in specific conditions to avoid this issue for the normal users.

joernalraun, 2022-11-28:
Hey @MarkusBordihn 
as you found out turning the matrix off does help. To put this into context (for others) when using pins from the display/led-matrix you have to make sure to turn the matrix off beforehand. This way you can use this pins as well for other tasks.