OpenRoberta/openroberta-lab #1363: Calliope Mini: No separated blocks for analog and digital pin operations

MarkusBordihn, 2022-09-07:
**Describe the bug**
Inside the robot configuration the digital and analog sensors and actors are separated, which makes sense because not all ports supports analog or digital signals.

[image]

But the corresponding blocks like write digital and write analog are not separated.
This creates confusions and makes it hard to document the single steps.

[image]

**To Reproduce**
Simple example for adding a digital read block you have to perform the following steps:
- Move the "get analog value" block to your program
- Click on the select box and change "analog value" to "digital value"
- Adjust the port to the corresponding block

**Expected behavior**
I would expect that analog and digital blocks are separated the same way like for the robot configuration to avoid any confusions for the user and to make it easier for the user to find the correct block for their use-case.

It's hard to explain why a kid needs first to move a "analog" block to their program and change it to "digital" later or the other way around.

For this reason my request is to create a separate block for analog and digital actors and sensors.

rbudde, 2022-09-15:
the same argument applies for micro:bit and arduinos, nano33ble and sensebox

rbudde, 2022-09-19:
done for the robots/embedded systems ardu, nano33bl3, sensebox, calliope, microbit. Will be deployed end of this week.

closed 2022-09-21 by rbudde