OpenRoberta/openroberta-lab #1367: 'disabled="true"' should NOT be used in our integtration tests

rbudde, 2022-09-22:
A lot of blocks are disabled in the integration tests. This can trigger a message "all blocks are used in tests", even if this is false.

```
Occurrences of 'disabled="true"' in Directory C:\git\openroberta-lab\OpenRobertaServer\src\test\resources\crossCompilerTests
Found Occurrences in Directory C:\git\openroberta-lab\OpenRobertaServer\src\test\resources\crossCompilerTests  (23 usages found)
    Usage in string constants  (23 usages found)
        OpenRobertaServer  (23 usages found)
            crossCompilerTests.robotSpecific.botnroll  (7 usages found)
                sensors.xml  (2 usages found)
                    519 <block type="robActions_display_text" id="6sIB9#hw2J8}V@AI6*^W" inline="true" disabled="true" intask="true">
                    535 <block type="robActions_display_text" id="sFGMC`6=m!UOVM6)1a6," inline="true" disabled="true" intask="true">
                text_colours_functions.xml  (5 usages found)
                    229 <block type="robActions_display_text" id="-O#A/x8u:3j?7dKaz.,+" inline="true" disabled="true" intask="true">
                    255 <block type="robActions_display_text" id="+;TJGyn1!P@dv]uvm*/C" inline="true" disabled="true" intask="true">
                    268 <block type="robActions_display_text" id=";~L(vbCJ*V/DS09R56;v" inline="true" disabled="true" intask="true">
                    281 <block type="robActions_display_text" id=",KT9j~vqjpw_xFB/1u9~" inline="true" disabled="true" intask="true">
                    294 <block type="robActions_display_text" id="ci%%|JtfW|fx;LDXFJTR" inline="true" disabled="true" intask="true">
            crossCompilerTests.robotSpecific.calliope  (1 usage found)
                math_lists.xml  (1 usage found)
                    834 <block type="mbedActions_display_text" id="IzWyo,`E_xK6*Py(Hr=c" disabled="true" intask="true">
            crossCompilerTests.robotSpecific.nano33ble  (1 usage found)
                aifes.xml  (1 usage found)
                    177 <block type="variables_get" id="bsNPn9,kp(RW27-aU75U" disabled="true" intask="true">
            crossCompilerTests.robotSpecific.nxt  (14 usages found)
                text_colours_messages_functions.xml  (14 usages found)
                    266 <block type="robActions_display_text" id="TRQiz3Y[{VEIoMqfrEtH" inline="true" disabled="true" intask="true">
                    285 <block type="robActions_display_text" id="U@:+w^]S^nm9uD{s57Cx" inline="true" disabled="true" intask="true">
                    323 <block type="robActions_display_text" id="MtjLVI!cMlIH*D9_hNV+" inline="true" disabled="true" intask="true">
                    342 <block type="robActions_display_text" id="}B}I!V2n-*|[^XB4*7%j" inline="true" disabled="true" intask="true">
                    361 <block type="variables_set" id="WT;cniJxAedgtu%1f(v)" disabled="true" intask="true">
                    370 <block type="variables_set" id="Lqdr.!u!!utZI(a@F?pZ" disabled="true" intask="true">
                    379 <block type="variables_set" id="R]GDhuL[DD=6+;hd+|VY" disabled="true" intask="true">
                    388 <block type="variables_set" id="pjb|0j*#vT+kW?ZWE0`~" disabled="true" intask="true">
                    397 <block type="variables_set" id="OY,vjKB:_S2Qe=I1hTxX" disabled="true" intask="true">
                    2475 <block type="robProcedures_defreturn" id="^HP8`}C.5nRbg0}*%p[f" disabled="true" intask="true">
                    2490 <block type="robProcedures_defreturn" id="Soy)n;Lpo+^]rqjMO=4v" disabled="true" intask="true">
                    2505 <block type="robProcedures_defreturn" id="a1|+rf/]afKbYT2CtpHS" disabled="true" intask="true">
                    2520 <block type="robProcedures_defreturn" id=":n{tbMj,kyj~3d!va;u/" disabled="true" intask="true">
                    2535 <block type="robProcedures_defreturn" id="tIT(~3[ZC6S7MsgmAxP2" disabled="true" intask="true">
```

rbudde, 2022-09-23:
unowifirev2 has disabled the RFID-RC522 blocks. If enabled, they give a warning and generate no code.

RaghuvirShirodkar, 2022-09-23:
Cross compilation fails for Uno Wifi Rev 2 on `develop`:

```
2022-09-23 19:23:15,259 INFO SPECIFIC-IT - [[[[[[[[[[ Robot: unowifirev2     compile file: /crossCompilerTests/robotSpecific/unowifirev2/action_1.xml
2022-09-23 19:23:15,263 INFO de.fhg.iais.roberta.util.Util - stored under: /tmp/dT_0C_bP_57c/prog/source/prog.cpp
2022-09-23 19:23:15,284 ERROR de.fhg.iais.roberta.util.Util - crosscompilation of NEPO generated program failed. Messages are logged to logger 'crosscompiler_error'
2022-09-23 19:23:15,284 INFO crosscompiler_error - 
***** cross compilation failed with response:
/tmp/dT_0C_bP_57c/prog/source//prog.cpp:8:10: fatal error: Adafruit_SSD1306.h: No such file or directory
 #include <Adafruit_SSD1306.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
***** for program:
// This file is automatically generated by the Open Roberta Lab.

#include <Arduino.h>

#include <Servo/src/Servo.h>
#include <LiquidCrystal_I2C/LiquidCrystal_I2C.h>
#include <Stepper/src/Stepper.h>
#include <Adafruit_SSD1306.h>
#include <NEPODefs.h>

...
...
...

void loop()
{
    ____action();
}
*****
2022-09-23 19:23:15,284 ERROR d.f.i.r.w.c.ArduinoCompilerWorker - compile unowifirev2 program prog failed with /tmp/dT_0C_bP_57c/prog/source//prog.cpp:8:10: fatal error: Adafruit_SSD1306.h: No such file or directory
 #include <Adafruit_SSD1306.h>
          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
2022-09-23 19:23:15,311 INFO SPECIFIC-IT - ---------- Robot: unowifirev2     FAILED (response-info) compile file: /crossCompilerTests/robotSpecific/unowifirev2/action_1.xml
2022-09-23 19:23:15,311 INFO SPECIFIC-IT - ]]]]]]]]]]
```

Missing library in `ora-cc-rsc` perhaps?

rbudde, 2022-09-26:
the Adafruit_SSD1306 block must be removed from the configuration, as it is not part of our distribution. If possible, the library should be added

bjost2s, 2024-10-23:
not time to work on it

closed 2024-10-23 by bjost2s