OpenRoberta/openroberta-lab #1368: ä will not be shown

Thorlogy, 2022-09-23:
**Describe the bug**
 - The letter "ä" will not be shown at the calliope (but the simulation is fine)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Calliope mini'
2. Click on block "show text"
3. Type a word with "ä"
4. See error

**Expected behavior**
 - Ä as small ä will be shown

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type Desktop
 - OS: windows
 - Browser chrome

**Additional context**
 - Add any other context about the problem here.

RaghuvirShirodkar, 2022-10-12:
Duplicate of #371 (?)

bjost2s, 2022-10-13:
This is a firmware problem/feature, you should create a ticket directly at Calliope / mbed

Thorlogy, 2022-10-13:
Would it be possible to show a note/hint that ä (umlaute) can not be shown on the Calliope mini? 
It should also be descriped in the help text:
[image]

closed 2022-10-13 by bjost2s