OpenRoberta/openroberta-lab #1369: dark light - calliope mini

Thorlogy, 2022-09-23:
**Describe the bug**
 - The dark led light will not be displayed (simulation is fine)
 - 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'calliope mini'
2. Click on block "turn led light on color" black
3. Download to calliope mini
4. See error

**Expected behavior**
 - LED is black

**Screenshots**
 - If applicable, add screenshots to help explain your problem.

**Device information**
 - Type desktop
 - OS: windows
 - Browser chrome

**Additional context**
 - Add any other context about the problem here.

RaghuvirShirodkar, 2022-10-12:
Black colour in RGB LEDs generally means that the three streams R + G + B = 0, so the LED remaining switched off _does indeed represent_  the colour black.

That being said, I suppose the simulation shows black simply because without it, it would appear as if the program did not run.

@bjost2s, @rbudde do you think we should consider this a simulation _error_ and correct it to maintain uniformity between the actual board and simulation?

bjost2s, 2022-10-13:
There are many more colors that a RGB LED cannot show:
[image]
["You can only make whatever color that is in the color wheel"](https://westsideelectronics.com/you-cant-make-brown-or-black-an-rgb-led-2/),
so brown and ather dark colors are also impossible. I think the simulation is fine, because we got the RGB value there and it can be displayed on a screen.

Thorlogy, 2022-10-13:
There should be a note within the Lab, that some colors can not be displayed on the Calliope mini. (The same hint/note will be shown sometimes when blocks can not be processed / used within the simulation.)

bjost2s, 2022-10-13:
We could make a remark in the block description, nothing else makes sense

closed 2022-10-13 by bjost2s

Thorlogy, 2022-10-13:
[image]

I guess the same should be possible for real robots

rbudde, 2022-10-14:
No, @bjost2s is right. The text shown in @Thorlogy 's example is generated at _compile time_ and states, that the block can never be used. It makes no sense to add a permanent note "some colors cannot be displayed", as you don't know at compile time, what colors are displayed. See the snippet atached. You can make it as complex as you want. BTW: the original claim, that black is not displayed on the real robot, is misleading. As @RaghuvirShirodkar noted, black means, that no color is shown on the LED.

A remark in the block description would be fine.

[image]