OpenRoberta/openroberta-lab #1391: Cannot execute create-empty-db on some Linux systems

chipbuster, 2022-10-20:
**Describe the bug**

When running `admin.sh create-empty-db` on Linux, database creation fails due to an exception thrown from `roberta.main.Administration`. This also means that a new server cannot be created, since the database creation done by `admin.sh start-server` will fail due to the same issue.

This issue is observed on at least two Linux machines, one running Ubuntu 18.04 and one running Arch Linux. A temporary workaround is to download an older version of ORA (5.0.4 or older).

I have bisected this issue on one of the affected systems. It appears to arise from commit adddbf2e113717f3f38ebb67ba2be2d60d26a131.

**To Reproduce**
Steps to reproduce the behavior:
1. Download openrobertalab_binaries.zip from [release 5.0.6](https://github.com/OpenRoberta/openroberta-lab/releases/tag/ORA-5.0.6)
2. Unzip release and `cd` into the  directory.
3. Run `./admin.sh create-empty-db`

Alternate reproduction:
1. Clone git repository
2. Run `mvn clean install -DskipTests`
3. Run `./admin.sh -git-mode create-empty-db`

**Expected behavior**
Database creation succeeds, allowing for the starting of a server.

**Screenshots**
 - If applicable, add screenshots to help explain your problem. [N/A]

**Device information**
 - Type: Desktop
 - OS: Ubuntu 18.04.6 LTS. Also observed on Arch Linux (rolling release).
 - Browser: Firefox 106.0, 64-bit

**Additional context**
Output of `java -version` on Ubuntu 18.04:

```
openjdk version "11.0.16" 2022-07-19
OpenJDK Runtime Environment (build 11.0.16+8-post-Ubuntu-0ubuntu118.04)
OpenJDK 64-Bit Server VM (build 11.0.16+8-post-Ubuntu-0ubuntu118.04, mixed mode, sharing)
```

Output of `java -version` on Arch Linux:
```
openjdk version "11.0.17" 2022-10-18
OpenJDK Runtime Environment (build 11.0.17+1)
OpenJDK 64-Bit Server VM (build 11.0.17+1, mixed mode)
```

<details><summary>Contents of `admin/logs/admin.log` after script failure</summary>

```
2022-10-20 01:48:05,652 [main] session-id=[] user-id=[] robot-name=[] INFO d.f.iais.roberta.main.Administration - *** create-empty-db ***
2022-10-20 01:48:05,671 [main] session-id=[] user-id=[] robot-name=[] DEBUG org.jboss.logging - Logging Provider: org.jboss.logging.Slf4jLoggerProvider
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.sun.xml.bind.v2.runtime.reflect.opt.Injector$1 (file:/home/chipbuster/tmp/openroberta-lab/OpenRobertaServer/target/resources/jaxb-impl-2.2.3-1.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int)
WARNING: Please consider reporting this to the maintainers of com.sun.xml.bind.v2.runtime.reflect.opt.Injector$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
2022-10-20 01:48:05,926 [MLog-Init-Reporter] session-id=[] user-id=[] robot-name=[] INFO com.mchange.v2.log.MLog - MLog clients using slf4j logging.
2022-10-20 01:48:05,932 [MLog-Init-Reporter] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.log.MLog - Reading VM config for path list /com/mchange/v2/log/default-mchange-log.properties, /mchange-commons.properties, /c3p0.properties, hocon:/reference,/application,/c3p0,/, /mchange-log.properties, /
2022-10-20 01:48:05,932 [MLog-Init-Reporter] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.log.MLog - The configuration file for resource identifier '/mchange-commons.properties' could not be found. Skipping.
2022-10-20 01:48:05,932 [MLog-Init-Reporter] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.log.MLog - The configuration file for resource identifier '/c3p0.properties' could not be found. Skipping.
2022-10-20 01:48:05,932 [MLog-Init-Reporter] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.log.MLog - The configuration file for resource identifier 'hocon:/reference,/application,/c3p0,/' could not be found. Skipping.
2022-10-20 01:48:05,932 [MLog-Init-Reporter] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.log.MLog - The configuration file for resource identifier '/mchange-log.properties' could not be found. Skipping.
2022-10-20 01:48:05,933 [main] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.cfg.MConfig - The configuration file for resource identifier '/mchange-commons.properties' could not be found. Skipping.
2022-10-20 01:48:05,933 [main] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.cfg.MConfig - The configuration file for resource identifier '/mchange-log.properties' could not be found. Skipping.
2022-10-20 01:48:05,933 [main] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.cfg.MConfig - The configuration file for resource identifier 'hocon:/reference,/application,/c3p0,/' could not be found. Skipping.
2022-10-20 01:48:05,933 [main] session-id=[] user-id=[] robot-name=[] DEBUG com.mchange.v2.cfg.MConfig - The configuration file for resource identifier '/c3p0.properties' could not be found. Skipping.
2022-10-20 01:48:05,990 [main] session-id=[] user-id=[] robot-name=[] INFO com.mchange.v2.c3p0.C3P0Registry - Initializing c3p0-0.9.5.5 [built 11-December-2019 22:18:33 -0800; debug? true; trace: 10]
2022-10-20 01:48:05,996 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.m.DynamicPooledDataSourceManagerMBean - MBean: com.mchange.v2.c3p0:type=PooledDataSource,identityToken=z8kfsxar1vqqjp11fqoipe|52fc5eb1,name=z8kfsxar1vqqjp11fqoipe|52fc5eb1 registered.
2022-10-20 01:48:06,001 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.m.DynamicPooledDataSourceManagerMBean - MBean: com.mchange.v2.c3p0:type=PooledDataSource,identityToken=z8kfsxar1vqqjp11fqoipe|52fc5eb1,name=z8kfsxar1vqqjp11fqoipe|52fc5eb1 unregistered, in order to be reregistered after update.
2022-10-20 01:48:06,001 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.m.DynamicPooledDataSourceManagerMBean - MBean: com.mchange.v2.c3p0:type=PooledDataSource,identityToken=z8kfsxar1vqqjp11fqoipe|52fc5eb1,name=z8kfsxar1vqqjp11fqoipe|52fc5eb1 registered.
[... 17 lines cut ...]
2022-10-20 01:48:06,482 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Test of PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT has SUCCEEDED.
2022-10-20 01:48:06,507 [main] session-id=[] user-id=[] robot-name=[] DEBUG d.f.i.r.persistence.util.DbExecutor - 1
2022-10-20 01:48:06,508 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Testing PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT.
2022-10-20 01:48:06,508 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Test of PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT has SUCCEEDED.
2022-10-20 01:48:06,508 [main] session-id=[] user-id=[] robot-name=[] DEBUG d.f.i.r.persistence.util.DbExecutor - 0
2022-10-20 01:48:06,509 [main] session-id=[] user-id=[] robot-name=[] DEBUG d.f.i.r.persistence.util.DbExecutor - 0
2022-10-20 01:48:06,510 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Testing PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT.
2022-10-20 01:48:06,510 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Test of PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT has SUCCEEDED.
2022-10-20 01:48:06,511 [main] session-id=[] user-id=[] robot-name=[] DEBUG d.f.i.r.persistence.util.DbExecutor - 1
2022-10-20 01:48:06,512 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Testing PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT.
2022-10-20 01:48:06,512 [main] session-id=[] user-id=[] robot-name=[] DEBUG c.m.v.c.i.C3P0PooledConnectionPool - Test of PooledConnection [com.mchange.v2.c3p0.impl.NewPooledConnection@4d1b54ed] on CHECKOUT has SUCCEEDED.
2022-10-20 01:48:06,513 [main] session-id=[] user-id=[] robot-name=[] DEBUG d.f.i.r.persistence.util.DbExecutor - 1
2022-10-20 01:48:06,513 [main] session-id=[] user-id=[] robot-name=[] ERROR d.f.i.r.persistence.util.DbUpgrader - no error: no db upgrades needed
[... cut ...]
javax.persistence.TransactionRequiredException: Executing an update/delete query
	at org.hibernate.internal.AbstractSharedSessionContract.checkTransactionNeededForUpdateOperation(AbstractSharedSessionContract.java:413)
	at org.hibernate.query.internal.AbstractProducedQuery.executeUpdate(AbstractProducedQuery.java:1668)
	at de.fhg.iais.roberta.main.Administration.createEmptyDatabase(Administration.java:142)
	at de.fhg.iais.roberta.main.Administration.run(Administration.java:96)
	at de.fhg.iais.roberta.main.Administration.main(Administration.java:69)
```
</details>

rbudde, 2022-10-20:
thank you very much. The problem: We updated the hibernate version to fix a security alert. In the new version there is a subtle API change: the shutdown (of the new created data base) needs a transactional context, but not in the old version. We deploy the change not later than Oct 25.

@chipbuster: your description was very helpful, especially the bisection. Thank you very much. I'd appreciate, if you test my changes next week and drop a note here, if everything is fine.

chipbuster, 2022-10-21:
@rbudde Thank you for the quick fix. I've tested branch `develop` on these systems, and the bug with `create-empty-db` appears to be fixed. I will be happy to test the release versions as well, when they come out.

Unfortunately, I'm encountering a different issue when trying to start the server: when I run `./admin.sh -git-mode start-server` and try to connect to it, I get an error 404.

[image]

I believe this is caused by line 130 in admin.sh hardcoding `server.staticresources.dir=./staticResources`, as this issue doesn't appear to show up in GitHub releases, and if I change the line to `server.staticresources.dir=./OpenRobertaServer/staticResources`, the script seems to work for the git repo. However, due to issues on my system, I am not confident in this diagnosis.

rbudde, 2022-10-21:
the admin script is used when starting the lab after exporting all resources. Locally we're using Intellij Idea and when working in 'git-mode' we don't use the script, but IDE-related stuff. So, we didn't encounter the error.

But, of course, you are right: it should work :-). I fixed it in commit 886fe1 .
Thank you very much, again. Hope, that everything works now. You are welcome to find more problems :-). It helps a lot!

chipbuster, 2022-10-22:
I have confirmed that everything is working on `develop` now. There were a few additional problems I ran into, but these were mostly due to me not reading the instructions properly.

Thank you again, and feel free to ping me if I forget to check the next release!

closed 2022-10-24 by bjost2s

chipbuster, 2022-10-26:
Confirmed that the 5.0.7 is working for me :)

rbudde, 2022-10-26:
@chipbuster: thanx. Hope, to hear from you again.