OpenRoberta/openroberta-lab #1392: xNN renaming output neurons

Thorlogy, 2022-10-20:
**Describe the bug**
 - A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'xNN Plugin'
2. Click on 'Neurnal Network NEPOnn'
3. Double-Click on output neuron 
4. See error

**Expected behavior**
 - The renaming window appears "far away" from the output neuron

**Screenshots**
[image]

**Device information**
 - Type: laptop
 - OS: Max
 - Browser chrome

rbudde, 2022-10-20:
a minor comment: please remove always the template text (as 'A clear and concise description of what the bug is.' resp. replace it.

Further: I expect, that the expected result is, that the rename window is *close* to the neuron.

rbudde, 2022-10-20:
getting the d3 relative mouse position behaves different for click and double click

Thorlogy, 2022-10-20:
It seems to work fine with Input-Neurons 
[image]