OpenRoberta/openroberta-lab #1399: Problems with invalid names for input/output neurons

rbudde, 2022-10-25:
Currently a name as "try" is marked as in valid. Problems are:
- the error message is attached to the start block. That is unexpected.
- there is no good reason, who "try" is forbidden (code generation adds an always safe prefix to the name
- renaming of an invalid name (sometimes!) fails. It is not obvious, what is the pattern of this behavior.
- the reason of the error is obvious, if you have more than 1 input/output neuron: blockly drop down issue

This is an old error and not introduced with the deployment of 5.0.7.
It is related to the issue #1348 and should be solved with this.
This error only hits user if they use names that are reserved words (checking this is a deprecated feature and must be removed)

rbudde, 2024-10-29:
I weakend the condition to be a valid neuron name. Then the problem vanishes.

bjost2s, 2024-12-11:
works as expected

closed 2024-12-11 by bjost2s