OpenRoberta/openroberta-lab #1403: Refactor the architecture of the IVisitor hierarchy

rbudde, 2022-10-28:
Adding a new plugin is much more complex as it should be. Especially the huge and hardly understandable IVisitor hierarchy is responsible for that. The architecture should be simplified drastically. This needs experiments to avoid new pitfalls :-)

**The final architecture should be:**
   - 1 interface in Robot declaring all visit*-methods for those blocks, that are common to ALL plugins (if, ..., ...) WITHOUT DEFAULTS
   - 1 interface for each plugin declaring the visit*-methods for blocks added in the plugin (sensor, actor, ...) WITHOUT DEFAULTS
   - all visit* methods in robot and all plugins should be final if possible. Clear Javadoc for each exception rom this rule
   
**goals for the far future:** 
1. ValidatorAndCollector and sub classes return BlocklyType instead of Void. The return type is the type
   of the blocks result, e.g. "if" -> Void; "max" -> Number; "and" -> Boolean
   This will be used later.
2. in the future we want to switch to a new blockly version. Adding especially sensors/actors should be very easy. All
   info about the blockly block, validation, code generation should be ideally at one place: the AST class.
   (goal: the refactoring is a prerequisite for that)
3. In many places the visit-Method in the ValidatorAndCollector-structure (e.g. "visitASTClass(ASTClass x)"
   can be removed and done generically. Idea:
   - the class annotation @NepoOp resp. @NepoPhrase have a flag "validateAndCollect=true or similiar.
   - in this case the the annotation processor doesn't call "visitASTClass(ASTClass x)",
     - it executes a "requiredComponentVisited(...)" call for all values
	 - and checks for an optional public method "addHardware(usedHardwareBuilder)". If the method exists,
	   it is called to update the hardware need of the program.
4. For simple cases this can be extended LATER to code generation

**steps (to be revised when done):**
- start with all sub-interfaces of IHardwareVisitor
- and remove them step by step (as long they don't have defaults, or???)
- then remove overrides by the regex:
	@Override\NL\ *public ==> public
- the result is:
  - tests should run
  - forgetting to add visit<X>(...)-method is not visible anymore - only in tests

- replace all public visit<X>(...)-methods by visit(...) (really? we can keep the visit* names to avoid overload problems)
- problems with tests
- but reuse... should run

rbudde, 2024-10-14:
the relevant changes mentioned above have been done. The "goals for the far future" will probably never implemented.

bjost2s, 2024-10-23:
solved in commits related to "textly"

closed 2024-10-23 by bjost2s