OpenRoberta/openroberta-lab #1408: updated RPi image

sce-taid, 2022-11-23:
https://www.open-roberta.org/lokale-installation/
currently links an outdated RPi image based on OR 4.2.1.

Please provide an updated RPi image based on OR 5.x.y.

Also it would be great if in addition to
https://github.com/OpenRoberta/openroberta-lab/wiki/Raspberry-Pi-2-3-4-and-the-Open-Roberta-Lab
you could describe the steps to (also) create a RPi image.

As an alternative a generic RPi image could be provided that pulls and runs the docker-based ORLab in a container.

bjost2s, 2022-11-28:
Unfortunately docker, raspi and the Open Roberta database do not work well together, so a docker - free installation is recommended. We do updates for the image from time to time. 
Please check if you can simply update the binaries from e.g. [here openrobertalab_binaries.zip](https://github.com/OpenRoberta/openroberta-lab/releases/download/ORA-5.0.9/openrobertalab_binaries.zip) on the latest image. Crosscompiler have not changed afaik

bjost2s, 2023-04-04:
I'll do my very best ;-)

bjost2s, 2023-04-14:
The Open Roberta Lab Raspi Image is now available. Please delete the cache if you have problems to download version 5.1.2. 
Please report issues here, if there are any ;-).

closed 2024-02-13 by bjost2s