OpenRoberta/openroberta-lab #1413: Change Robotino serial write block

heini208, 2022-12-02:
**Is your feature request related to a problem? Please describe.**
 - Currently the output of the serial write block for the Robotino is not really usable as it requires deep understanding about how the implementation functions and even then it's very cumbersome to access. 
 -  
**Describe the solution you'd like**
- A potential alternative could be to save the printed text into a variable which can then be watched inside the Robotino Web-interface.
- Depending on how the Web-interface displays variables, the implementation could either just show the last print or show every print thus far.
- Since even that is not very intuitive to use but very useful for implementation and debugging of features, the serial write block could be removed from the expert toolbox and the Debug block (CNTRL-2) could be used instead.

**Describe alternatives you've considered**
 - If the above idea does not work, the block should still be removed as it's basically unusable

heini208, 2022-12-23:
To make a program available to the web-interface, all you have to do is put it into the directory /home/robotino/programs. This can either be done via ssh or the rest API with a POST to /data/uploadProgram.

Starting and stopping a program is done by using this function in the web-interface program.html with a PUT request:
with data being an object that resembles the program to be started/stopped:

`function actionCellClick(e, cell)
{
  var row = cell.getRow();
  var data = row.getData();
  
  console.log(data);
  
  //gtable.clearData();
  //otable.clearData();
  
  $.ajax({
		url: "data/startStopProgram?sid=" + RobotinoRESTsession,
		type: 'PUT',
		data: JSON.stringify(data),
		contentType: 'application/json'
	});
}`

[image]
Data for the view-program "Circle-Kreis" which is the first program in the list (id=1)

bjost2s, 2024-10-24:
no time to work on it

closed 2024-10-24 by bjost2s