OpenRoberta/openroberta-lab #1417: Radio blocks in Calliope mini blue Robot

joernalraun, 2022-12-05:
When using Calliope mini blue as a robot and you build a script with radio blocks it can't be compiled.
"Das Programm oder die Konfiguration enthält Fehler. Überprüfe die Meldungen."

It make sense not to compile a script with radio blocks selected when using a mobile device (with a Bluetooth connection) as the Bluetooth-stack is running on the same frequency as the radio function.
However, as users can select it, the error message is not correct as it states that the program ist not correct (although it is).

A solution could be a correct error message clarifying the issue or no appearance of radio blocks when Calliope mini blue is selected (this would make more sense as it is a uncommon behavior when you can select functions, but in the end the lab tells you that this feature can't be used).

heini208, 2023-03-01:
I have the following ideas to make the Warning more generally applicable: 

1. There are warnings in your program or configuration. _Please check the messages_.
2. There are warnings in your program or configuration. _Please address the warnings_.
4. It seems a block in your program or configuration is preventing _the program_ from starting. Please check the messages.
5. It seems a block in your program or configuration is preventing _it_ from starting. Please check the messages.
7. (Oops!) The program can't run because of a warning. This could be caused by a block in the program or the program's configuration. Please address all warnings before using the program.

joernalraun, 2023-03-03:
>  have the following ideas to make the Warning more generally applicable

Hello heini208,
thanks for this suggestion, but having it generally doesn't help here as it is unclear why the warning appears.
I haven't seen a message yet, if there is any it might be ok, but so far you would get an error referring to a message that hasn't appeared. From my experience this is too complicated for a user to follow.

heini208, 2023-03-03:
Hello @joernalraun 
first, thank you for the feedback.
[image]
This Error Message you see popup when trying to open the code view needs to be generally applicable. There is currently no option to make it specific to each error. Also, there can be multiple errors in the same program, which could lead to confusion if only one pops up or too much pops up.

A more specific Error message that can be adjusted for each case should pop up on the block causing the error.
In this case, that would be the following message:

[image]

Do you think this needs to be adjusted instead?
Maybe to "This version of the robot does not support this block!"

joernalraun, 2023-03-03:
Thanks for your images! If you get this error message it will be very helpful.

heini208, 2023-03-03:
I think this might be a different issue entirely. Because the picture above is what currently happening in the LAB. 
But I just realized it is very easy to miss if you move the program around.

Moving the Program makes the error disappear. This is especially annoying if you are using it on mobile, so in a bigger program you cannot really search for the error message.

@joernalraun 
Could this be correct? Are you using mobile by any chance and missing the error message?
Or do you think only the text needs to be adjusted?

heini208, 2023-03-10:
This ticket will be closed and discussion is moved to #1463.

closed 2023-03-10 by heini208