OpenRoberta/openroberta-lab #1428: Robotino Simulation: endless loop when using "drive ... distance cm"-block

PamoCurtis, 2022-12-13:
**Describe the bug**
The robot does not stop driving when setting the distance value of the **'drive x speed %, y speed %, distance cm'**-block **(robActions_motorOmni_curve_for)** to zero. No other blocks will be executed after this.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to dev.open-roberta.org and select Robotino
2. Import [robotinoDriveDistanceBug.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/10219517/robotinoDriveDistanceBug.xml.txt) **(you need to remove the '.txt' first)**
3. Open and start the simulation
4. See error

**Expected behavior**
Since the distance is set to zero, the robot shouldn't move and it might be good if a warning is displayed that the distance should not be zero

RaghuvirShirodkar, 2023-03-07:
The issue seems to be partly fixed, as the robot does seem to move a little bit even when distance = 0:
 
[image]

In the `System Values` on the right, one can see the `Robot X` values changing.

rbudde, 2023-03-14:
works as expected

closed 2023-03-14 by rbudde