OpenRoberta/openroberta-lab #1434: Mbot2: Open Roberta Connector programm fails while uploading a program to mbot2

bjost2s, 2022-12-30:
This happens on older laptops in most cases, about 8 out of 10 cases. This issue cannot be reproduced on e.g. my linux machine but on other windows laptops it happens more or less often depending on the machine.

RaghuvirShirodkar, 2023-03-09:
Tested source code in `openroberta-connector`. Uploading the program is retried in total 3 times (max) now, after which an error is returned. Function definition can be [found here](https://github.com/OpenRoberta/openroberta-connector/blob/0699c61338b96926766ce461cca35bebf705c6aa/src/main/java/de/fhg/iais/roberta/connection/wired/mBot2/Mbot2Communicator.java#L216).

RaghuvirShirodkar, 2023-03-14:
Tested on mBot2 with the test server. Upload works as expected.

closed 2023-03-14 by bjost2s