OpenRoberta/openroberta-lab #1435: Spike Prime / Robot Inventor Integration

bjost2s, 2023-01-02:
Integrate robot [Spike Prime / Robot Inventor](https://education.lego.com/en-gb/products/lego-education-spike-prime-set/45678#spike%E2%84%A2-prime) in the OpenRoberta Lab.

- [ ] Check robot capabilities
- [ ] Add code generation (Micro Python)
- [ ] Add configuration (differential drive)
- [ ] Add tests
- [ ] Simulation
- [ ] Add background image
- [ ] Add configuration image
- [ ] Add icon
- [ ] Add wiki entries/help files
- [ ] Add program transfer possibilities in the [Connector Program](https://github.com/OpenRoberta/openroberta-connector)

bjost2s, 2023-02-09:
forgotten issue linking 14c90adbb60498f270b1e3b0e6f060a723b629a4

rbudde, 2023-03-14:
link to LEGO couls not be supplied due to legal reasons. Works as expected.

closed 2023-03-14 by rbudde