OpenRoberta/openroberta-lab #1439: Edison: try / catch not supported

bjost2s, 2023-01-05:
The following program [EdisonExceptionTest.xml.txt](https://github.com/OpenRoberta/openroberta-lab/files/10354031/EdisonExceptionTest.xml.txt) cannot be tranferred to edison, the reason is the not supported try / catch statement:
[image]

Maybe there is an option to work with newer/better firmware, this should be checked before trying to solve this bug. The solution might not be easy to implement.

bjost2s, 2023-02-06:
fixed in 081d23be14c9fdee9947640c4626de0b657b132a, together with #1446

RaghuvirShirodkar, 2023-03-03:
Tested successfully on test server. Toolbox does not contain `break out`/`continue with next iteration` block for Edison.

closed 2023-03-08 by bjost2s