OpenRoberta/openroberta-lab #1455: Introduce "deprecated" property for plugins / systems

bjost2s, 2023-02-15:
Over the years we have integrated a couple of systems and some of them are / become outdated now and we would like to remove them from our platform (and repo)

To inform the users we should introduce a new property "deprecated" connected with the following actions:

- label the deprecated system with a stamp in the start popup, similar to "beta"
- react to the user choise for a deprecated system with a warning popup, explaining what happens and providing information for workarounds (alternative systems, how to update, etc). This warning should appear each time the deprecated system is chosen.
- explaining reasons and workarounds in our [wiki](https://wiki.open-roberta.org)

RaghuvirShirodkar, 2023-03-03:
Tested the following points on [test server](https://test.open-roberta.org/):

- [X] label the deprecated system with a stamp in the start popup, similar to "beta".
- [X] react to the user choise for a deprecated system with a warning popup, explaining what happens and providing information for workarounds (alternative systems, how to update, etc). This warning should appear each time the deprecated system is chosen.
- [ ] explaining reasons and workarounds in our [wiki](https://wiki.open-roberta.org/).

There seems to be a new "bug" now. The icons for the robots in the start carousel are shifted to the right/not centered:
[image]

Maybe related to this issue?

rbudde, 2023-03-14:
works as expected

closed 2023-03-14 by rbudde