OpenRoberta/openroberta-lab #1461: Micro:bit v2 Implementation

heini208, 2023-03-01:
This ticket tracks the progress for the implementation of the updated micro:bit v2

A summary regarding new microbit v2 features can be found [here](https://support.microbit.org/support/solutions/articles/19000119052-details-of-micro-bit-v2#comparison)

### New microbit v2 Features:

The microbit v2 supports the following [new features](https://makecode.com/blog/microbit/microbit-V2): 

**An upgraded Speaker supports new speaker features:**
- Turning the speaker on and off
- Playing new sounds: giggle, twinkle, yawn

**New microphone:** 
- if Loud/Quiet Sound
- Set Loud/Quiet Threshold
- Get Sound level (0-255)

**New Touch Logo:**
- if logo pressed?

**Pin Touch Modes:**
- The pins (P0, P1, P2) and the logo can be set to capacitive touch or resistive touch.
- resistive means: you need to touch the pin and the ground, capacitive means you only need to touch the pin

### New File format:
This is also discussed in ticket #949

WIP

heini208, 2023-03-03:
To generate a HEX-file usable on Microbit currently uflash is being used.
There is an updated version of uflash (2.0.0) that creates a "universal hex file".
The version currently used in the lab is 1.2.4 and does not support universal hex files.

uflash could be updated and the compile.py used by the [MicrobitCompilerWorker.java](https://github.com/OpenRoberta/openroberta-lab/blob/dcfeba5e442a829266ed26cd13fee46fb4d8f624/RobotMbed/src/main/java/de/fhg/iais/roberta/worker/MicrobitCompilerWorker.java#L33) can be adjusted (the new compile.py is already done [here](https://github.com/OpenRoberta/ora-cc-rsc/tree/feature/microbitV2)).

This would mean that the hex file created in the CompilerWorker now supports both Microbit and MicrobitV2. This is also very easy to implement as it only requires a minor adjustment in the ora-cc-resources and an update of the uflash library.

One Major disadvantage of this approach is the size of universal hex files.
A Universal Hex file has a estimate size of ~1.8mb while the hex file currently created for the microbitv1 was only ~650kb this is a major increase.

One approach could be to only use the Universal Hex file for the new microbit. However the newer uflash version does no longer support any other hex type besides the universal one. So updating uflash from 1.2.4 to 2.0.0 causes the original compile.py of the microbitv1 to no longer work. 

Something that could be further explored is the javascript library by microbit [microbit-fs](https://microbit-foundation.github.io/microbit-fs/) which seems to support non-universal hex files.

PamoCurtis, 2023-04-17:
Issues Simulation:
 - "get LED brightness" does not work (cannot read properties of undefined (reading 'pixel') interpreter.robotSimBehaviour.js:420)
 -

PamoCurtis, 2023-04-17:
Issues:
- Configuration error when using pin 0 -> "Another component is already using the same port!"
- Block "mbedActions_display_getPixel" does not work in the simulation(cannot read properties of undefined (reading 'pixel') -> __interpreter.robotSimBehaviour.js:420__)
- Block "robSensors_accelerometer_getSample" is present twice in the expert toolbox
- Simulation isn't loading when using sensor configuration block and using a pin > 2 (cannot set properties of undefined (setting 'name') -> __robot.microbitv2.js:107__ and __robot.microbitv2.js:114__)
- Blocks "mbedCommunication_sendBlock", "mbedCommunication_setChannel",  "actions_play_file", "actions_play_expression" and "actions_sound_toggle" are not usable in the simulation, which is correct, but they do not stop the simulation from opening and executing

closed 2023-05-08 by bjost2s